arduino-libraries/Arduino_OV767X #2: Error compiling for board Arduino NANO 33 IoT

baqwas, 2020-07-06:
Hello,

Unable to compile Camera Test Pattern sketch. The error messages are:

Arduino: 1.8.12 (Linux), Board: "Arduino NANO 33 IoT"
Arduino: 1.8.12 (Linux), Board: "Arduino NANO 33 IoT"

WARNING: library Arduino_OV767X claims to run on mbed architecture(s) and may be incompatible with your current board which runs on samd architecture(s).
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp: In member function 'void OV767X::beginXClk()':
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:330:3: error: 'NRF_I2S' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
   ^~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:330:3: note: suggested alternative: 'ID_I2S'
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
   ^~~~~~~
   ID_I2S
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:330:28: error: 'I2S_CONFIG_MCKEN_MCKEN_ENABLE' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:330:61: error: 'I2S_CONFIG_MCKEN_MCKEN_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:330:61: note: suggested alternative: 'I2S_CLKCTRL_MCKEN_Pos'
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                             I2S_CLKCTRL_MCKEN_Pos
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:331:29: error: 'I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2' was not declared in this scope
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:331:68: error: 'I2S_CONFIG_MCKFREQ_MCKFREQ_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:331:68: note: suggested alternative: 'I2S_CLKCTRL_MCKEN_Pos'
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                    I2S_CLKCTRL_MCKEN_Pos
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:332:26: error: 'I2S_CONFIG_MODE_MODE_MASTER' was not declared in this scope
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:332:26: note: suggested alternative: 'USB_CONFIG_POWERED_MASK'
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                          USB_CONFIG_POWERED_MASK
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:332:57: error: 'I2S_CONFIG_MODE_MODE_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:332:57: note: suggested alternative: 'SYSCTRL_BOD33_MODE_Pos'
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
                                                         SYSCTRL_BOD33_MODE_Pos
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:334:24: error: 'digitalPinToPinName' was not declared in this scope
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
                        ^~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:334:24: note: suggested alternative: 'digitalPinToBitMask'
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
                        ^~~~~~~~~~~~~~~~~~~
                        digitalPinToBitMask
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:334:57: error: 'I2S_PSEL_MCK_PIN_Pos' was not declared in this scope
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
                                                         ^~~~~~~~~~~~~~~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:334:57: note: suggested alternative: 'I2S_CTRLA_CKEN_Pos'
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
                                                         ^~~~~~~~~~~~~~~~~~~~
                                                         I2S_CTRLA_CKEN_Pos
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp: In member function 'void OV767X::endXClk()':
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:342:3: error: 'NRF_I2S' was not declared in this scope
   NRF_I2S->TASKS_STOP = 1;
   ^~~~~~~
/home/reza/Arduino/libraries/Arduino_OV767X/src/OV767X.cpp:342:3: note: suggested alternative: 'ID_I2S'
   NRF_I2S->TASKS_STOP = 1;
   ^~~~~~~
   ID_I2S
exit status 1
Error compiling for board Arduino NANO 33 IoT.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Board information:
BN: Arduino Nano 33 BLE
VID: 0x2341
PID: 0x805a
SN: 5BC0CE9072E5BEAF

baqwas, 2020-07-06:
Not an issue. Operator/user error. Sorry.
Selected the wrong board!

closed 2020-07-06 by baqwas

HenFo, 2024-01-14:
Hi, I have the same issue, but I use a Nano 33 IoT... Are the components incompatible?

Board Info:
BN: Arduino NANO 33 IoT
VID: 0x2341
PID: 0x8057
SN: 93E8E43050304D48502E3120FF170C2A

c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp: In member function 'void OV767X::readFrame(void*)':
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:236:1: error: 'NRF_GPIO_Type' was not declared in this scope
 NRF_GPIO_Type * port;
 ^~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:236:1: note: suggested alternative: 'TCC_PER_Type'
 NRF_GPIO_Type * port;
 ^~~~~~~~~~~~~
 TCC_PER_Type
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:236:17: error: 'port' was not declared in this scope
 NRF_GPIO_Type * port;
                 ^~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:236:17: note: suggested alternative: 'Port'
 NRF_GPIO_Type * port;
                 ^~~~
                 Port
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:238:10: error: 'nrf_gpio_pin_port_decode' was not declared in this scope
   port = nrf_gpio_pin_port_decode(&ulPin);
          ^~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp: In member function 'void OV767X::beginXClk()':
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:361:3: error: 'NRF_I2S' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
   ^~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:361:3: note: suggested alternative: 'ID_I2S'
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
   ^~~~~~~
   ID_I2S
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:361:28: error: 'I2S_CONFIG_MCKEN_MCKEN_ENABLE' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:361:61: error: 'I2S_CONFIG_MCKEN_MCKEN_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:361:61: note: suggested alternative: 'I2S_CLKCTRL_MCKEN_Pos'
   NRF_I2S->CONFIG.MCKEN = (I2S_CONFIG_MCKEN_MCKEN_ENABLE << I2S_CONFIG_MCKEN_MCKEN_Pos);
                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                             I2S_CLKCTRL_MCKEN_Pos
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:362:29: error: 'I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2' was not declared in this scope
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:362:68: error: 'I2S_CONFIG_MCKFREQ_MCKFREQ_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:362:68: note: suggested alternative: 'I2S_CLKCTRL_MCKEN_Pos'
   NRF_I2S->CONFIG.MCKFREQ = I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV2  << I2S_CONFIG_MCKFREQ_MCKFREQ_Pos;
                                                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                    I2S_CLKCTRL_MCKEN_Pos
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:363:26: error: 'I2S_CONFIG_MODE_MODE_MASTER' was not declared in this scope
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:363:26: note: suggested alternative: 'USB_CONFIG_POWERED_MASK'
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                          USB_CONFIG_POWERED_MASK
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:363:57: error: 'I2S_CONFIG_MODE_MODE_Pos' was not declared in this scope
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:363:57: note: suggested alternative: 'SYSCTRL_BOD33_MODE_Pos'
   NRF_I2S->CONFIG.MODE = I2S_CONFIG_MODE_MODE_MASTER << I2S_CONFIG_MODE_MODE_Pos;
                                                         ^~~~~~~~~~~~~~~~~~~~~~~~
                                                         SYSCTRL_BOD33_MODE_Pos
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:365:24: error: 'digitalPinToPinName' was not declared in this scope
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
[... 1 lines cut ...]
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:365:24: note: suggested alternative: 'digitalPinToBitMask'
   NRF_I2S->PSEL.MCK = (digitalPinToPinName(_xclkPin) << I2S_PSEL_MCK_PIN_Pos);
                        ^~~~~~~~~~~~~~~~~~~
                        digitalPinToBitMask
[... cut ...]
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:373:3: error: 'NRF_I2S' was not declared in this scope
   NRF_I2S->TASKS_STOP = 1;
   ^~~~~~~
c:\...\Arduino\libraries\Arduino_OV767X\src\OV767X.cpp:373:3: note: suggested alternative: 'ID_I2S'
   NRF_I2S->TASKS_STOP = 1;
   ^~~~~~~
   ID_I2S

Using library Arduino_OV767X at version 0.0.2 in folder: C:\...\Arduino\libraries\Arduino_OV767X 
Using library Wire at version 1.0 in folder: C:\...\Arduino15\packages\arduino\hardware\samd\1.8.13\libraries\Wire 
exit status 1

Compilation error: exit status 1