arduino-libraries/Arduino_OV767X #5: Test Pattern zigzag

hpssjellis, 2020-09-30:
@facchinm @8bitkick 

Can someone confirm that this version 0.0.2 of the arduino ov7670 library runs properly with the Nano 33 BLE and the [Omnivision ov7670 camera. ](https://www.amazon.com/CANADUINO-Camera-Module-Compatible-Arduino/dp/B07S99V4TN/ref=sr_1_1?dchild=1&keywords=CANADUINO%20Camera%20Module%20Compatible%20with%20Arduino%20-%20OV7670%20Image%20Processor%20-%20640x480%20VGA%20Resolution&qid=1601274522&sr=8-1&fbclid=IwAR36x6d1Mn1vw5jH4XPX7ZWSi7CYWVc30wEO7sTXOmcjjswct-hfu6ll91I)

For me it generates a zigzag pattern (using Processing version 3.5.4  https://processing.org/download/support.html ) with board manager mbed nrf5284 version 1.1.6 for the Nano 33 BLE. Note that loading older versions of this board does not seem to help. 

[image]

.

The ov7670 library does not generate any pattern with arduino_mbed_enabled boards version 1.2.2 for the Nano 33 BLE/PortentaH7.

This blog was probably made for version 0.0.1 ov7670 release (When I tried to install version 0.0.1 it did not install, when I downloaded it from this github it did not work). https://blog.arduino.cc/2020/06/24/machine-vision-with-low-cost-camera-modules/

It seems other people are having the same issue at this arduino forum   https://forum.arduino.cc/index.php?topic=697294.0

I assume I have done something wrong but I am getting consistent bad results with multiple different wirings etc.  Here is my camera view  

[image]

Please reply. It would be interesting to know what is working for you.

facchinm, 2020-09-30:
Hi @hpssjellis ,
the pins have been completely swapped between 0.0.1 and 0.0.2 to improve the acquisition speed (at the cost of slightly more complicated wiring).
This is the commit introducing the change https://github.com/arduino-libraries/Arduino_OV767X/commit/8480d6653b07ba77827596b288c906096b8cfa3a#diff-dae1d9b78874283b76d32664b92d34ffL12

The tutorial in fact refers to 0.0.1 . 

@per1234 @dcuartielles @8bitkick should we change the table to refer to the new pin assignment? Or maybe duplicate it to allow 0.0.1 users to wire it correctly?

hpssjellis, 2020-09-30:
@facchinm I am using the new wiring and I get the zigzag pattern. Both images are using the new wiring.

mbanzi, 2020-09-30:
I'm using 0.0.2 and I get the correct pattern. the Processing sketch loses sync after a while. that needs to be investigated further

hpssjellis, 2020-09-30:
Thanks so much for looking into this @mbanzi .  

I am having better luck with version 0.0.1 of the ov7670 library. The color seems washed out with the 0.0.2 library. (Yes I could have a wire messed up but I have connected these several times.) Out of curiosity how long are your cables? Mine are regular 8 inch female to female cables. 

[image]

per1234, 2020-09-30:
>should we change the table to refer to the new pin assignment?

@facchinm I think that's the best thing to do. If you follow the tutorial instructions, you get the latest version of the library, so the table should match that version.

>Or maybe duplicate it to allow 0.0.1 users to wire it correctly?

I think that adds too much complexity to the tutorial. At the very most, I might add a note at the bottom:
>**Note:** this wiring is for the latest version of the Arduino_OV767X library. If you're using v0.0.1, follow the wiring in the comment at the top of the example sketch.

But I don't think it's very likely people reading the tutorial will decide to select "0.0.1" from the Library Manager's version menu, so I don't think that's necessary.

---
You could side-step the issue by replacing the table with the instructions:
>Connect the wires according to the comment at the top of the example sketch.

But this would require a reordering of the tutorial:
1. hardware requirements
1. install library
1. open example sketch
1. hardware setup
1. upload example sketch
1. ...

I much prefer the current hardware -> software order, and the table is pretty nice!

hpssjellis, 2020-09-30:
By the way. I have fixed my color issue using verion 2 of the library 0v7670 library by flipping pins D0 and D1. Not sure why you guys aren't having the same issue.  

See this commit.   
[https://github.com/arduino-libraries/Arduino_OV767X/commit/977bb7096ca7566c5b951f3cb158134172f0fbf3](https://github.com/arduino-libraries/Arduino_OV767X/commit/977bb7096ca7566c5b951f3cb158134172f0fbf3)

[image]

I think the zigzag issue is the length of my wires. Do your test setups have wires that are 8 inches long?

8bitkick, 2020-09-30:
> 
> I think the zigzag issue is the length of my wires. Do your test setups have wires that are 8 inches long?

Thanks @hpssjellis - Per @mbanzi's comment the issue is likely down to lost bytes on the application side. what Processing version are you running? Are you using windows or Mac?

hpssjellis, 2020-10-01:
> Thanks @hpssjellis - Per @mbanzi's comment the issue is likely down to lost bytes on the application side. what Processing version are you running? Are you using windows or Mac?

I use Windows, good point. Before I change my wires I think I will try Processing on Ubuntu since it looks like it runs on Linux. [https://processing.org/download/](https://processing.org/download/)

hpssjellis, 2020-10-01:
You guys rock @mbanzi and @8bitkick. I just needed to use Ubuntu. Looks good. I might still mess around with it on windows and see if there is anything I can do.

[image]

hpssjellis, 2020-10-01:
Above image was 1 FPS, I tried 5 FPS and QQVGA and it did not work on Ubuntu, so I changed the Serial rate to 115200 and 5 FPS worked great. 

[image]

I also tried serial 115200 on windows and it was a huge improvement I got the occassional good picture with some zigzag randomly. (I still think wire length is part of the issue, but I don't have any female-female short wires, was thinking of making them, but would probably just mess up a lot of wires.)

[image]

hpssjellis, 2020-10-05:
I think my camera module is wired slightly differently than yours.

So I tested short wires and it has zero effect on the zigzag pattern using  windows. I did prove that my board is reversed D0/D1 than your wiring. (colors work if I switch D0/D1) They go from pink/green to normal. Is it possible my board has something else switched which only effects the zigzag?  

[my Cam](https://www.amazon.com/Tulead-OV7670-Arduino-Compatible-Control/dp/B07XX2RM2C/ref=sr_1_47?dchild=1&keywords=Camera+Module+for+Arduino+%E2%80%93+OV7670+Image+Processor+%E2%80%93+VGA+640%C3%97480&qid=1601934091&sr=8-47)

If that is a possibility can someone make a suggestion about which pins if switched might improve the situation. I would think the VS and HS pins.

hpssjellis, 2020-10-06:
I think this site has interesting information at the [bottom of the page](https://wiki.beyondlogic.org/index.php/OV7670_Camera_Module_with_AL422_FIFO_Theory_of_Operation) about 2 versions of the ov7670 with version 2 having the HREF on a different pin!

8bitkick, 2020-10-06:
Thanks - yes I have seen a few inconsistencies in pinout between modules which is confusing things a bit. We are working on a viewer which should fix the zig zag

8bitkick, 2020-10-06:
One thing to check @hpssjellis -with the color corrected what result are you seeing from [ConnectionTest.ino](https://github.com/arduino-libraries/Arduino_OV767X/blob/master/examples/ConnectionTest/ConnectionTest.ino)

hpssjellis, 2020-10-06:
ConnectionTest.ino seems to give the same set of errors as before, but a new viewer would be great. Eventually I will need some kind of organized serial viewer of the data, but I can probably make that. The end result is putting it into a Machine Learning program like [this small one](https://www.rocksetta.com/tensorflowjs/beginner-keras/27keras-medium-image.html) or [this larger one](https://www.rocksetta.com/tensorflowjs/beginner-keras/28keras-full-image.html). Being able to minupulate the resolution and color data would be very helpful. I am looking into [OV7670app.pdf](http://web.mit.edu/6.111/www/f2015/tools/OV7670app.pdf) but it is a bit over my head. Thanks for the interest in this.

[image]

hpssjellis, 2020-10-07:
Since the issue seems to be with Java on Windows I am not going to worry about it anymore. However I will still look into changing the resolution. I did find this useful code. Which should go well with the manual.

[https://manualzz.com/doc/13427058/ov7670](https://manualzz.com/doc/13427058/ov7670)

```

from https://forum.arduino.cc/index.php?topic=535819.0

#define CAMERA_ADDRESS 0x21

byte readCameraRegister(byte registerId) {
  Wire.beginTransmission(CAMERA_ADDRESS);
  Wire.write(registerId);
  Wire.endTransmission();
  Wire.requestFrom(CAMERA_ADDRESS, 1);
  while (Wire.available() <= 0);
  byte registerValue = Wire.read();
  delay(1);
  return registerValue;
}

void writeCameraRegister(byte registerId, byte registerValue) {
  Wire.beginTransmission(CAMERA_ADDRESS);
  Wire.write(registerId);
  Wire.write(registerValue);
  Wire.endTransmission();
  delay(1);
}

```

closed 2021-03-09 by mbanzi