arduino-libraries/Arduino_OV767X #6: Feature request: Example for the PortentaH7

hpssjellis, 2020-09-30:
Please design an ov7670 library example for the Arduino PortentaH7. I am doing a fair bit of examples and am only interested in the ov7670 library to be able to port it to the Portenta if possible. My other examples are at 

[https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7)

If you can make an attempt at the example I would be fine testing it as soon as I get female headers for my Portenta.

@sbhklr

sbhklr, 2020-09-30:
@hpssjellis Might be just a matter of mapping different ports. I'll investigate.

hpssjellis, 2020-12-27:
Using the new release mBED core 1.3.1  the CameraCaptureRawBytes Demo does not compile.

facchinm, 2020-12-28:
@hpssjellis to support the Portenta a patch must be done at library level and is not trivial (furthermore, it would make little sense unless you use the hardware camera interface which is only located on high density connectors). I'd avoid bitbanging but if it's something useful I can try coding something.

hpssjellis, 2020-12-28:
It would be very useful.

@facchinm yes the high density connector does make sense and from talking with the [facebook Arduino zoom group](https://www.facebook.com/groups/portenta/permalink/690146031639942/) with @sebromero and others,  I think a color camera shield is on it's way (no idea when), however just getting a color camera working is only the first step. Tons of machine learning programs could use a color camera, using Arduino C/C++ code and the use of the inner M4 core, but that also will not be trivial to get working. 

[My Arduino Machine Learning](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/m09-Tensoflow)

[My TensorflowJS Machine Learning](https://www.rocksetta.com/tensorflowjs/)

The [ov7670 low cost](https://www.amazon.com/HiLetgo-OV7670-640x480-0-3Mega-Arduino/dp/B07S66Y3ZQ/ref=sr_1_4?dchild=1&keywords=ov7670+module+universal+soldier&qid=1609170458&sr=8-4) camera solution working with the PortentaH7 would get a huge head start going on Standard Color Machine Learning using the Portenta. The OpenMV solution has a few issues (grey scale only, single core only, micro-python only)

hpssjellis, 2021-01-17:
It looks like the Portenta Vision Shield examples are getting updated. I got a Machine Learning program working with the updates but are still stuck with the Grayscale that the ov7670 camera would solve. Presently the O7670 camera is not working with my Machine Learning using edge Impulse becuase I think the frameBuffer is not being zeroed, like they are doing with the Portenta framebuffer camera. Any susggestions for zeroing the frame buffer.

[https://github.com/arduino/ArduinoCore-mbed/pull/122](https://github.com/arduino/ArduinoCore-mbed/pull/122#issuecomment-761617418)

closed 2022-02-17 by hpssjellis