arduino-libraries/Arduino_OV767X #8: Feature Request. Any chance of QQQVGA Resolution?

hpssjellis, 2020-11-06:
Any chance of getting a few more resolutions out of the OV7670 Camera module. Here is a list of what OpenMV has going, For a bit of machine learning I would just like a few sizes smaller than what we already have, such as: 

QQQVGA: 80x60
QQQQVGA: 40x30

Note: Smaller is faster and easier to process.

Data from [OpenMV](https://docs.openmv.io/library/omv.sensor.html) 

sensor.QQCIF: 88x72
sensor.QCIF: 176x144
sensor.CIF: 352x288
sensor.QQSIF: 88x60
sensor.QSIF: 176x120
sensor.SIF: 352x240
sensor.QQQQVGA: 40x30
sensor.QQQVGA: 80x60
sensor.QQVGA: 160x120
sensor.QVGA: 320x240
sensor.VGA: 640x480
sensor.HQQQVGA: 80x40
sensor.HQQVGA: 160x80
sensor.HQVGA: 240x160
sensor.B64X32: 64x32 (for use with image.find_displacement())
sensor.B64X64: 64x64 (for use with image.find_displacement())
sensor.B128X64: 128x64 (for use with image.find_displacement())
sensor.B128X128: 128x128 (for use with image.find_displacement())
sensor.LCD: 128x160 (for use with the lcd shield)
sensor.QQVGA2: 128x160 (for use with the lcd shield)
sensor.WVGA: 720x480 (for the MT9V034)
sensor.WVGA2:752x480 (for the MT9V034)
sensor.SVGA: 800x600 (only for the OV2640/OV5640 sensor)
sensor.XGA: 1024x768 (only for the OV2640/OV5640 sensor)
sensor.SXGA: 1280x1024 (only for the OV2640/OV5640 sensor)
sensor.UXGA: 1600x1200 (only for the OV2640/OV5640 sensor)
sensor.HD: 1280x720 (only for the OV2640/OV5640 sensor)
sensor.FHD: 1920x1080 (only for the OV5640 sensor)
sensor.QHD: 2560x1440 (only for the OV5640 sensor)
sensor.QXGA: 2048x1536 (only for the OV5640 sensor)
sensor.WQXGA: 2560x1600 (only for the OV5640 sensor)
sensor.WQXGA2: 2592x1944 (only for the OV5640 sensor)

lightb0x, 2020-12-11:
`64x64` and `96x96` support would be lovely for ML applications.
Transfer learning from [imagenet_resized](https://www.tensorflow.org/datasets/catalog/imagenet_resized), [pretrained MobileNet V2](https://github.com/tensorflow/models/tree/master/research/slim/nets/mobilenet) will be possible.

hpssjellis, 2020-12-30:
Looking at the code it doesn't seem like it would be that hard to do QQQVGA or QQQQVGA
Code sample [here](https://github.com/arduino-libraries/Arduino_OV767X/blob/30668622f7e98c4fd248ee005a6bdcee37948b10/src/OV767X.cpp#L91-L94)

This looks like the main entry point for QQVGA 160 x 120 
QQQVGA  80 x 60
QQQQVGA  40 x 30

```
    case QQVGA:
      _width = 160;
      _height = 120;
      break;

```

facchinm, 2020-12-30:
@hpssjellis that is the easy part :slightly_smiling_face: 
Then the sensor must be configured with the right resolution using `ov7670_configure` (since the nRF52 has no hardware crop, Portenta has and uses it). Formars are then converted using these tables (https://github.com/arduino-libraries/Arduino_OV767X/blob/30668622f7e98c4fd248ee005a6bdcee37948b10/src/utility/ov7670.c#L835-L892); most of the values are empirically determined since Ominivision datasheets are simply filled with errors and incorrect information.
So, it's a big effort. Anyway, if anyone is willing to take it I'll be glad to support.

mbanzi, 2021-01-19:
BTW there is a new pull request which adds a number of resolutions to the library, check it out [here](https://github.com/arduino-libraries/Arduino_OV767X/pull/12 )

closed 2021-03-09 by mbanzi