arduino-libraries/Arduino_OV767X #9: Why does this require mbed enabled boards?

jwestmoreland, 2020-12-08:
Hello All,

Can the library be modified to work with non mbed enabled boards like the Nano 33 IoT?

I'll take a look at the lib when I have time - and I'll make a pull request if I get it to work.

Thanks,
John W.

mbanzi, 2021-01-19:
Hi John. It can be modified to work with other boards  but this library was specifically designed for the Arduino Nano 33 BLE / BLE Sense for a number of reasons.  

- It is meant to be used in Tiny ML applications and you need at least a Cortex M4 for decent performance 
- The camera needs a clock source and we are currently generating that using a pin. Currently that part of the code is highly dependent on the Nano 33 BLE. 

It is possible to use a Nano 33 but mostly to take single snapshot. Somewhere I have the code to generate the clock with a SAMD21 but we currently don't have the bandwidth to make that modification.

The "processing sketch" is provided with the library to visualise the incoming images on a computer. (the computer needs to be quite fast otherwise the image will look mangled )
sorry we haven't had the time to answer before

jwestmoreland, 2021-01-19:
Hello Massimo,

If you want to send me the code for that - I'll take a look and if I get it to work, I'll post it - and do a PR for it.

Taking snapshots right now would be OK with me.

No worries - thanks for responding.

Best Regards,
John W.

mbanzi, 2021-01-26:
I can'f ind my code but I remember I derived it from this post from user MartinL on the Arduino Forum https://forum.arduino.cc/index.php?topic=672239.msg4529804#msg4529804

I changed the parameters in  DPLLRATIO to generate around 10Mhz and it worked. (the code example currently generates 32MHz)

mbanzi, 2021-01-26:
I'm going to close this now. If somebody comes back with a PR to support the SAMD21 we're happy to look at it :) thanks!!

closed 2021-01-26 by mbanzi

jwestmoreland, 2021-01-26:
Hello Massimo,

OK - I'll take a look at that and give it a go soon - NP on closing this.

If it works I'll do a PR,

Thanks,
John